lukh/frameforge #10: Obtenir un profil non attaché.

VinceBa, 2024-10-19:
Une possibilité a été supprimé: celle d’obtenir un profil non attaché avec "create profil"
Il suffit de sélectionner un profil avec les paramètres sans sélectionner de segment d'attachement et de valider.
L'utilisateur peut ainsi placer le profil à sa guise ou l'utiliser pour autre chose.

lukh, 2025-01-22:
j'ai réimplémenté sur develop, j'ai séparé la spinbox "length" du reste des options comme elle est pas pour moi dans le groupe des options "width, height, fillet, etc).

C'est soit la longueur soit les edges, il reste à empecherl'ui de laisser croire que l'on peut faire les deux

lukh, 2025-01-22:
et il faudrait empecher le "ok" tant que lenght est nul ou que pas de edge selected

lukh, 2025-03-10:
Done, still on develop though

closed 2025-03-10 by lukh

VinceBa, 2025-03-16:
great.