lukh/frameforge #11: Introduction des courbes de tuyauterie

VinceBa, 2024-10-19:
Une fonction à ajouter à l'atelier serait la création de courbes (ou coudes) de tuyaux à souder. C'est très utilisé en tuyauterie et chaudronnerie pour faire des structures ou pour tuyauterie industrielle. Dans la pratique on prends une courbe de 90° et on la coupe à la demande pour relier deux tuyaux, puis on soude le tout. Dans 90% des cas on sert de 90° ou 45°. C'est là que l'atelier peut intervenir: créer un un coude normalisé. Il suffirait de détecter l'angle entre deux edges et de le placer à l’intersection et correctement orienté.
Ensuite on peut rajouter les tuyaux en prenant les surfaces des coudes et faire un loft (ou extrude).
Un peu dans l'idée mais sans avoir l'arc de cercle: [viewtopic.php?t=90482&start=20](https://forum.freecad.org/viewtopic.php?t=90482&start=20)

lukh, 2025-08-28:
french duplicate of #42

SergioP1975, 2025-08-29:
I think that the insertion of standard accesories (flanges, elbows, tees, reducers...) should be left to the internal Assembly workbench. There has to be a way to create library of comercial parts (not only pipe accesories but also fastenings or any comercial item) that could be integrated in the internal Assembly workbench (not anoter workbench, integrated in Assembly directly). I see several third parties workbenchs that deals with the same problem in different ways (fasteners, gears, Dodo/Quetzal...) that could be solved by a unified integrated library.

And the same hapens with the frame workbenchs, there are several ones (in my opinion FrameForge the only really functional by the moment), but there are other similars that could be integrated in the FrameForge (cables/wires, pipes, anything that is a section extruded along a path)

closed 2026-10-12 by lukh