lukh/frameforge #9: Utilisation de profils divers

VinceBa, 2024-10-19:
Une possibilité intéressante serait d'utiliser des profils fournis par l'utilisateur. Soit dans un repertoire sur le disque, soit déjà introduite dans le dessin en cours d'édition.
Peut être dans le choix "material" avoir une option "select face" et "select file".
Ce choix "material" pourrait alors s'appeler "source".
Il y aurait alors le choix:
- Profilés normalisés acier
- Profilés normalisés aluminium
- Profilé depuis sélection
- Profilé depuis fichier
- Poutre bois (éventuellement)
Exemple d'utilisation pour les profilés aluminium divers et variés utilisé pour des chassis d’appareil électrique/électronique.
[PG20.zip](https://github.com/user-attachments/files/17445910/PG20.zip)

MichaelPanzer, 2025-03-11:
Has there been any progress on this? I think it would be a very useful feature.

lukh, 2025-03-16:
[image]

teasing... ;)

lukh, 2025-03-16:
on alu-extrusions branch for now.

I used scripts from BOLTS.

I need to find a (clever) way to add custom profile. The issue is how to add that to the UI...

MichaelPanzer, 2025-03-17:
Do you have a link to that branch? I cant seem to find it.

As far as custom profiles I'm pretty sure both OnShape and SW allow you to define custom profiles in using sketches in other files. I'm not sure how difficult that would be to implement here but I think it would make the best workflow for people coming over from other CAD packages.

lukh, 2025-03-21:
not yet pushed, got a problem, 

It seems it is not possible to save "freecad shapes or object", through the pickle process that save/reload data from the python class.

So not useable for now since it is not reloaded correclty. To Be Continued...

Maybe the right way to do it is to save a string with filename/object name and find it through the API.

lukh, 2025-11-16:
You all wanted custom profiles, here is a custom profile:

[image]

The Aluminium extrusion is also almost working. gotta merge that this week end maybe

VinceBa, 2025-11-17:
I can't do. Extrude must be done from a wire... like standard wires... because they will come from dxfs.
[ProfilesALU_from_DXF.FCStd.txt](https://github.com/user-attachments/files/23591427/ProfilesALU_from_DXF.FCStd.txt)

closed 2026-01-06 by lukh