lukh/frameforge #22: Changing all the frame sizes

sdifru, 2025-01-23:
I made cube frame. It is made of rectangular hollows. They have the size of 60x40x2.5 each. I want to change all the rectangular hollow sizes to 50x50x2.5. How do I do it?

lukh, 2025-01-23:
For the moment it is not possible,
You have to delete the existing frame and recreate it.

I'll keep the ticket open, should be implemented.

lukh, 2025-03-11:
Well. It turns out it is possible, with one quirk.
Select the profiles you want to modifyin the tree view, and change their params, for instance height and width, or whatever you want.

[image]

They will be updated. 

Two things: you need to manually set the params, ie, no "profile database" for set/defined params.
If the size is specified in the name (new feature of 0.1.3), then it will not be updated.

Still, it should be added, I think it is some work; delaying it

lukh, 2025-03-15:
You can have a look at branch "edit-profiles".

I'd love to have feedback :)

sdifru, 2025-03-20:
I've tried the edit-profiles branch. It did open a task where I could change the size of the profile. It changed only the size of the double-clicked profile. How do I change the size of multiple profiles?

Did you change the edit-profiles branch's name? I've tried to download it again but I couldn't because it didn't exist. @lukh

lukh, 2025-03-21:
It has been merged on develop, and removed.

I don't see a way (regarding the current implementation i mean) to do that. TBC