lukh/frameforge #27: Link to object go out of the allowed scope

sdifru, 2025-03-06:
I opened frameforge workbench. I created a part and a sketch in it. After finished drawing, I created profile on it. When it generated hollow, freecad showed this message:
`Part::FeaturePython: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Profile_Rectangular_Hollow_60x40x005'. Instead, the linked object(s) reside within 'Part'.`

When I changed a parameter of the hollow, the similar messages showed again.

lukh, 2025-03-06:
Known issue.
After using it for a while, the workflow explained in the tutorial is not right. 
Creating a Part, adding a sketch in it and then create profiles for the sketch's edge will lead to this issue. I think it is because the Origin of the profile is not the same as the one from the part... 

You can drag and drop the profiles in the part, but that will lead with more errors after that (when trimming it.)

The proposed workaround for now is the following: 

Create the sketch directly in the root of the document; create the profiles and trim them. when you are happy, create a Group, or a fusion for it. it will group all the profile inside a subfolder( i personnally use a lot the fusion, since it allows a basefeature and partdesign on it)

Correcting the bug (ie, allowing to create the profile where you want will be handled, the tutorial will be updated as ell)

lukh, 2025-03-10:
Should be  fixed, on develop for now

closed 2025-03-10 by lukh