lukh/frameforge #35: Profile sections become invisible after cut operation

Perolet, 2025-06-19:
When I perform a cut on the profiles, they visually disappear from the screen.
It seems that the cut causes the profile elements not to be displayed, even though they are still present in the model.

[image]

Perolet, 2025-06-21:
It fixed itself. Sorry for the inconvenience.

Perolet, 2025-06-22:
The error has occurred again: when I make a section cut, the pipes disappear from the model. Could you tell me what I might be doing wrong or if there’s a setting I should check?

[image]

Perolet, 2025-07-09:
Thanks again for your reply.

I understand, of course, that you're not obligated to help — and I truly appreciate your time.

That said, since the response took a while to arrive, I was hoping for a slightly more in-depth look at the issue. The explanation you gave seems to address a very basic scenario, which I believe doesn’t apply here — the task is already completed, and as shown in the images, the corners are as they should be.

I’d be happy to share the file in case it helps clarify what’s going on. There may be something more subtle causing the behavior.

Again, thank you for your time and support.

lukh, 2025-07-14:
Please share the file yes :)

lukh, 2025-07-16:
I'll check to fix this, keeping it open; but at least there is a "fix"

raorn, 2025-07-27:
[file.zip](https://github.com/user-attachments/files/21456563/file.zip)

Try to milter-cut TrimmedProfile015 and TrimmedProfile018. Also, the following error appears:

```
00:45:51  Recompute TrimmedProfile028
00:45:51  pyException: Traceback (most recent call last):
  File "/home/raorn/.local/share/FreeCAD/Mod/FrameForge/freecad/frameforge/trimmed_profile.py", line 106, in execute
    normal = Part.Plane(p1, p2, p3).toShape().normalAt(0,0)
                        ^^
<class 'UnboundLocalError'>: cannot access local variable 'p1' where it is not associated with a value
00:45:51  Recompute TrimmedProfile029
00:45:51  pyException: Traceback (most recent call last):
  File "/home/raorn/.local/share/FreeCAD/Mod/FrameForge/freecad/frameforge/trimmed_profile.py", line 106, in execute
    normal = Part.Plane(p1, p2, p3).toShape().normalAt(0,0)
                        ^^
<class 'UnboundLocalError'>: cannot access local variable 'p1' where it is not associated with a value
00:45:51  TrimmedProfile028: cannot access local variable 'p1' where it is not associated with a value
00:45:51  TrimmedProfile029: cannot access local variable 'p1' where it is not associated with a value
```

raorn, 2025-07-28:
> Try to milter-cut TrimmedProfile015 and TrimmedProfile018 (I DO select surfaces from 3d view, tried different pairs from different angles). Also, the following error appears:

This is me being an idiot. Sketch004 is missing horisontal size constraint and two parts were not aligned by 0.14mm

> File "/home/raorn/.local/share/FreeCAD/Mod/FrameForge/freecad/frameforge/trimmed_profile.py", line 106, in execute
>     normal = Part.Plane(p1, p2, p3).toShape().normalAt(0,0) [...]

There should be `else` branch in condition above that gives error "parts are not aligned" or something like that.

trexxet, 2025-08-04:
Yep, selecting from 3d view works for now, though I find using the tree more convenient.

lukh, 2026-02-23:
Another possible error : #76

closed 2026-02-23 by lukh