lukh/frameforge #36: codebase doesn't adhere to PEP8, making contributions difficult.

z0r0, 2025-07-14:
Hello,

As a developer I would like to contribute to the frameforge project, however it is difficult to do so, mostly due to the fact that the codebase doesn't implement many python standards, conventions, or best-practices.

These standards are defined [here](https://peps.python.org/pep-0008/) and are implemented in various tools in the python landscape. They are meant to enable collaboration, make code easier to read and ultimately make contribution and collaboration easier. 

I have created a PR which integrates, and applies two tools (isort, and black) which are standard tools, that automatically apply PEP8 to your codebase. [See the PR here](https://github.com/lukh/frameforge/pull/34)

My goal is to actively enable more contributions to this project, including my own.

lukh, 2025-08-11:
Hi @z0r0,
Since it took me ages to come back on that, i decided to work on it the easiest way.
I didn't merge #34  because: 
- reverted commit, and several changes in between, 
- AND i'd like to keep the "main" branch  stable and release only. I'd like to merge all PR on "develop", which should be stable. 

I took the liberty to update develop (via a linting branch) with your change, and a little bit more (a project section able to install isort and black, for the CI)

z0r0, 2025-09-01:
I'll be honest, this is quite frustrating. These commits are pretty simple. I rolled out black and isort, as well as created a pyproject.toml file, I reverted the changes, made some tweaks to the pyproject.toml file, and re-ran isort and black.

The changes made by isort and black make no logical changes, they only make linting changes. Please read the documentation, OR comment in the PR. 

My intention is to work with you collaboratively on your project, but in order to do that, we've got to work on good foundations of this project, standardization, and unit tests before any of the needed refactoring is able to take place.

lukh, 2025-09-04:
done :)